lhr-solar/BPS #76: Voltage OpenWire functions should reference system_open_wire field in cell_asic struct

SijWoo, 2020-06-05:
The Voltage functions that handles checking open wires are using the return value of the LTC6811_run_openwire_multi function. The return value is not scalable and could overflow thus missing certain open wired pins. There are problems when having multiple minion boards daisy chained as using the LTC6811_run_openwire_multi function. Errors arise in Voltage_GetModulesInDanger because of that ltc function. There is a system_open_wire field which uses a bitmap where each index corresponds to a pin and a 1 in that index will indicate that there is an open wire. The OpenWire functions should be modified to use the system_open_wire field in the cell_asic minions struct instead of the return value of the LTC6811 function.

**TODO:** Make the Voltage OpenWire functions reference system_open_wire field when checking for open wires.
_Note:_ Use the Voltage test if you are testing it.

SijWoo, 2020-06-09:
Start on this after simulator_spi_bsp has been merged

ClarkPoon, 2020-07-09:
@SijWoo was this fixed by the **Voltage Library cell_asic fix** pull request?

SijWoo, 2020-07-11:
This has not been fixed yet. The **Voltage Library cell_asic fix** branch was fixed a different issue.

closed 2020-10-17 by ClarkPoon