lhr-solar/BPS #83: Simulator: Locking files during read and writes

SijWoo, 2020-06-09:
The BSP src code and the simulator python modules need to lock the file that they are accessing or else errors will occur. Make sure to add the following in your codes:

Python File Accesses:
```python
import fcntl
import os
```

```python
os.makedirs(os.path.dirname(file), exist_ok=True)
    with open(file, 'r') as csvfile:
        fcntl.flock(csvfile.fileno(), fcntl.LOCK_EX)    # Lock file before writing
        # Do stuff
        fcntl.flock(csvfile.fileno(), fcntl.LOCK_UN)    # Unlock file after writing
```

BSP C File Accesses:
```C
// Open File and get file descriptor
FILE* fp = fopen(file, "r");
if(!fp) {
    // Exit, error!
    perror("SPI.csv");
    exit(EXIT_FAILURE);
}

// Lock the file so simulator.py/SPI.py can not write it during a read op
// This is a blocking statement
flock(fp, LOCK_EX);

// File reading stuff

// Unlock the lock so the simulator can write to SPI.csv again
flock(fp, LOCK_UN);

// Close file.
fclose(fp);
```

**TODO:** Make sure all file accesses have locks

SijWoo, 2020-06-09:
BSP_ADC ne

chaseblock, 2020-06-27:
What needs to be done in order to close this issue?

SijWoo, 2020-06-27:
All python scripts and C src files that are currently finished in the simulator branch should have locks. The ADC related src files need to have locks and a couple other. Let’s close this issue whenever every BSP src files and python scripts that are currently in the simulator branch have locks then make sure every future pull request have locks.

closed 2020-07-04 by ClarkPoon