lhr-solar/BPS #98: Python config file that contains file path locations

SijWoo, 2020-06-27:
`file = "BSP/Simulator/DataGeneration/Data/FILE.csv"` is used in a lot of python scripts but if we decide to change the directory path or a file name, we have to go through each script and edit that. Moving this to a central location and having individual variables for the directory and file names will make editing easier and cleaner.

**TODO:** Create a config python script that holds file names/locations and make all the python scripts that are currently in the simulator branch access these variables.

**Future Note:** Make sure all future merges use the config file.

ClarkPoon, 2020-06-28:
I just realized that I added more file that have this problem when I merged in the pull requests for PLL and Watchdog

chaseblock, 2020-07-05:
Is this issue ready to be closed with the merge of #105 into simulator?

SijWoo, 2020-07-05:
Strobelight.py isn’t using config.py. I’ll update that

closed 2020-07-09 by SijWoo