lhr-solar/BPS #102: Simulator runs C code before the "hardware" has started

ClarkPoon, 2020-06-28:
Right now, the code running on the BPS Simulator is executed before the BeVolt has been started, which is not realistic. Ideally, the code running on the BPS simulator will be executed after BeVolt has been started.

chaseblock, 2020-06-30:
To confirm: is the issue here that we would prefer for the C code to be executed only after giving the `start` command to the python simulator program?

ClarkPoon, 2020-06-30:
Yes, that is correct. Otherwise we will run into bugs/race conditions. For example, the python script sets the clock to 16MHz on startup, but usually the C code will set the clock to 80 MHz on startup. Every time I have run this on the simulator, C will set the PLL first, but in the actual BPS, C will set the PLL after the hardware sets the PLL

PR #107 (2020-07-03): Coordinated launch
Fix the issue of the BeVolt software launching before its hardware is ready.

Closes #102.
closed 2020-07-03 by chaseblock