lhr-solar/BPS #113: STM32F413: Update BSP_LED GPIO pins for new leader board v3.0

SijWoo, 2020-07-18:
To make the board look cleaner, I moved around some GPIO pins that controls the LED. (In case anyone is wondering, the low speed communication lines such as UART, I2C, SPI have not been changed). The updated pin mapping of the leader board is in the [google sheet](https://docs.google.com/spreadsheets/d/1CHXOJur96z3ZEMzSynHYqkd0Sx3a4DRUVOYeTzj7Trg/edit?usp=sharing).

The new LED pin assignments are as followed:
PC0 : Other
PC1 : CANError
PC2 : WDError
PC3 : (Open) Wire
PC4 : RUN

PA4 : OCurr
PA5 : OTemp
PA6 : OVolt
PA7 : UVolt

Do not branch off the simulator branch for this. I'll create a note whenever the leader board layout is done and merged to the Electronics repo in case you need to reference it.

chaseblock, 2020-07-18:
Can we maintain this and a separate build target for a v2 board? I imagine that we will be testing code on the old master for at least a little while.

SijWoo, 2020-07-18:
Ah that would be good, so have separate folders like BSP/STM32f413v2 and BSP/STM32F413v3 or with some type of version number naming scheme?

chaseblock, 2020-07-18:
Nah that sounds pretty ugly. I was thinking just a file that has variables/definitions that are guarded by preprocessor #ifs.

Something like:
```c
// Some .h file somewhere
#ifdef BPS_V2
// etc.
#define CANError_LED_PIN ...
#define WDError_LED_PIN ...
// etc.
#else
// etc.
#define CANError_LED_PIN PC1
#define WDError_LED_PIN PC2
// etc.
#endif
```

Then just select the build target in the Makefile by passing in `-DBPS_V2` or `-DBPS_V3` to gcc when compiling.

I'm not sure what sort of variables would need to be switched, so the feasibility of this may vary, but it sounds more maintainable than separate BSP folders (since the BSPs are likely going to be similar to each other).

SijWoo, 2020-07-18:
Okay yea that sounds better. I can edit the Makefile to have a version parameter. I'll open a separate issue ticket for the Makefile.

SijWoo, 2020-08-01:
Will put this on hold for now. Look at issue [#114](https://github.com/lhr-solar/BPS/issues/114) for more information

closed 2020-12-12 by ClarkPoon