lhr-solar/BPS #114: Update Makefile to have version parameter

SijWoo, 2020-07-18:
There are multiple versions of the BPS boards with different pin connections. The Makefile needs to be updated to accept a version parameter so src files for different board versions can be compiled.

The Makefile will accept a version number as a parameter, a variable will be defined for GCC to select the preprocessor definitions for the certain board, then the code will be compiled.

Src code example:
```
// Some .h file somewhere
#ifdef BPS_V2
// etc.
#define CANError_LED_PIN ...
#define WDError_LED_PIN ...
// etc.
#else
// etc.
#define CANError_LED_PIN PC1
#define WDError_LED_PIN PC2
// etc.
#endif
```

SijWoo, 2020-07-30:
I just realized since we're using a different current board with different I/O interfaces, how should we change the Current (Amperes) library to support different versions (ADC and AS8510)? The new current sensing board is using the AS8510 which requires the SPI protocol so the Current library will be using the AS8510 driver whenever it gets created. Should we have the current library have a bunch of #if statements to support the previous and newest leader board versions or should we just have the code updated to only support the new version and just deprecate the previous versions?

I think having a bunch of #if statements in the Current library will create clutter quickly and we need to change the Current library function definitions completely when we update the code to use the AS8510. I think we should deprecate the previous version of the bps but have it available **ONLY** for testing certain modules that remains unchanged (e.g. Voltage, Temperature, Contactor, CAN, USB, EEPROM). LEDs and Strobe Light are the only pins that have been changed on the new board. We'll have at least two boards (before making multiple copies) to test the critical functionalities.

To save the BPS code that uses the hall effects sensors + ADC, we can create a tag called v2.0 like what we did for v1.0.

Any thoughts are welcome :)
@ClarkPoon @chaseblock @sugamarora99 @manthanand @alekxoo @AleNeff @etblum12

SijWoo, 2020-08-01:
After discussing with BPS team, we decided to add the makefile version parameter after adding the AS8510 driver and editing the amperes library. The main concern that I had with the makefile version paremeter was that the amperes library will be different between the new and previous leader boards. As long as we don't test code that is meant on the new leader board i.e. the fan, amperes board, bluetooth on the previous version, then we'll be fine for now.

ClarkPoon, 2021-02-10:
To my knowledge, we no longer have any functional copies of old versions of the BPS leader board (@chaseblock took a DC-DC converter off of the board from Spring 2020 to use on the telemetry board), so I no longer see the point of adding support for legacy hardware.

closed 2021-02-10 by ClarkPoon