lhr-solar/BPS #121: Control-C not working properly on simulator

ClarkPoon, 2020-07-27:
When I run the simulator on my machine and press Control-C, then type quit, the simulator freezes instead of quitting. When Alejandro runs the simulator on his machine and he presses Control-C, the simulator immediately quits (without opening the quit menu) Not sure if these issues are related, but they were both first noticed today after the pull requests for changing SoC to Charge and for Contactor Simulation were merged into the simulator branch.

ClarkPoon, 2020-07-27:
After looking into this further, it appears that Alejandro is no longer having this issue on his machine. The issue on my machine appears to be caused by the Timer thread not ending when the "quit" option is selected if the C code is not using the timers

chaseblock, 2020-07-27:
On my machine (Pop!_OS 20.04), I seem to get the expected results (no issue with quitting).

ClarkPoon, 2020-07-27:
> On my machine (Pop!_OS 20.04), I seem to get the expected results (no issue with quitting).

Can you try running after building with 
```
make simulator TEST=BSP_I2C
```
I am not encountering this issue when I build with
```
make simulator
```

chaseblock, 2020-07-27:
Yes that seems to do it. I can confirm that issue.

PR #123 (2020-07-28): Fixed the edge case where the timer would not die if it had not been …
This should fix the issue where control-C, then "quit" would not successfully quit the simulator if the timers had not been started yet.

Closes #121
closed 2020-07-28 by chaseblock