lhr-solar/BPS #128: Current library to Amperes library?

SijWoo, 2020-07-30:
Even though this has been what we've been using for the past couple years, every time I say current board/library with other members I feel compelled to specify Current means amperes and not time. Current members know current means amps and not time in the context of the BPS but for future members, it may be good to change the library name to prevent any confusion.

**Proposal:** Change Current library and Current board names to Amperes library and Amperes board

SijWoo, 2020-07-30:
Any thoughts are welcome :)
@ClarkPoon @chaseblock @sugamarora99 @manthanand @alekxoo @AleNeff @etblum12

ClarkPoon, 2020-07-30:
I think this is a good idea

chaseblock, 2020-08-02:
I'm not too much of a fan of this move, but I understand the difficulty with the current "current" term, and as long as this improves clarity, I don't mind.

closed 2020-12-14 by ClarkPoon