lhr-solar/BPS #168: Updating standard peripheral library to low-layer API

SijWoo, 2020-08-29:
STM released a library called Low-Layer(LL) API after the standard peripheral library(SPL). They're both similar in syntax, but STM has been supporting LL and deprecated the SPL. **Would this be something that you all would want to do?** I don't know if you all can allocate more members to this or not but wanted to let you all know! If you do want to do this, I can send the library src folder to you.

Just as a note, LL is different from STM's High Abstraction Layer(HAL) library which has some overhead. There may not be a lot overhead in HAL, but there are some register level stuff we need that HAL can't provide easily.

chaseblock, 2020-08-29:
There's an overview of HAL and LL here: https://www.st.com/content/ccc/resource/technical/document/user_manual/2f/77/25/0f/5c/38/48/80/DM00122015.pdf/files/DM00122015.pdf/jcr:content/translations/en.DM00122015.pdf

From what I can tell, LL does not provide a CAN API. Since this is somewhat critical to our operation, I could see this being an issue. We could write our own drivers at the register level instead, but that seems like a non-trivial project.

SijWoo, 2020-08-29:
Yea that would be an issue. I did not see that, thanks for catching that Chase. We're able to have a hybrid of HAL's CAN API and LL on a project with no conflicts. The UART and SPI peripherals are the ones where we need to be close to the registers level (for the BPS with RTOS at least). What do you all think about this option?

JimothyGreene, 2020-08-29:
Why exactly do we need to be close to register level with those peripherals? Not sure I fully understand this. Haven't been dealing this low level recently.

SijWoo, 2020-08-29:
We need those peripherals on interrupts for the RTOS to block/unblock tasks. Peripheral data transfers are blocking functions, so the OS can schedule a different task to run if the original task is waiting on a peripheral transfer to be completed. In order to unblock a task though, we need an indicator for when a transfer is complete. The interrupt handler can be triggered whenever a transfer is complete and inside the handler we can call a semaphore to unblock the task.
E.g. for the BPS, we have one task for voltage/temperature monitoring and the other task is for amperes monitoring (volt/temp task priority higher than amperes task priority). While the LTC driver is doing an SPI data transfer in the volt/temp task, the amperes monitoring task can running. Whenever the SPI transfer is complete, the SPI interrupt handler will unblock the volt/temp task. The OS will then schedule the volt/temp task to run now.

```
OS_SEM SPI_Sem4;

void VoltTempTask(void) {
    while(1) {
        Voltage_UpdateMeasurements();   // Blocking statement because LTC driver calls SPI1_Transfer
        // Do stuff

        Temperature_UpdateMeasurements();   // Blocking statement because LTC driver calls SPI1_Transfer
        // Do more stuff
    }
}

void AmperesTask(void) {
    while(1) {
        Amperes_UpdateMeasurements();
        // Do stuff
    }
}

void SPI1_Transfer(uint8_t data[], uint32_t len) {
    for(int32_t i = 0; i < len; i++) {
        OSSemWait(&SPI_Sem4);
        SPI1_DATA_R = data[i];
    }
}

void SPI1_IRQHandler(void) {
    // Ack tx complete
    OSSemPend(&SPI_Sem4);
}
```

SijWoo, 2020-08-29:
Just to note, that code is an example and not actually in the BPS XD

JimothyGreene, 2020-08-29:
I think using the mixed HAL and LL option would be good.

ClarkPoon, 2021-02-10:
I think this would be a lot of work, and not worth the effort, since our BSP already works (for the most part)

closed 2021-02-10 by ClarkPoon