lhr-solar/BPS #192: [RTOS] General: Make sure all functions are thread-safe

SijWoo, 2020-08-31:
More specific issue tickets should/will be created but as a note, we need to make sure all the functions are thread safe (at least the ones that are shared among multiple tasks).

chaseblock, 2021-03-20:
This issue has been standing for a long time, and I'm of the opinion that it needs some modification due to decisions made over the past few months. Not every function needs to be 'thread-safe'. Instead, perhaps we should take the stance that only certain functions need to be thread safe, and that each function should be marked as either `[reentrant](https://en.wikipedia.org/wiki/Reentrancy_(computing))`, `[thread-safe](https://en.wikipedia.org/wiki/Thread-safety)`, or `unsafe`. This could be specified plainly in the documenting comment for the function.

For quick description:
- A thread-safe function is something that can be called from multiple threads at the same time, and can be called at the same time as other thread-safe functions in the same module.
- A reentrant function is one that (for our purposes) would be safe to call from multiple threads, as well as from interrupts.
- An unsafe function (again, for our purposes) is neither of the above.

The hope is that going through all of the functions and labeling them as one of these categories would make it clear whether we have race-conditions, potential deadlocks, etc. We would need to go through and ensure all of our driver and BSP functions are categorized appropriately.

Looking for criticism here. Let me know what y'all think.

@ClarkPoon @JimothyGreene

ClarkPoon, 2021-03-20:
> This issue has been standing for a long time, and I'm of the opinion that it needs some modification due to decisions made over the past few months. Not every function needs to be 'thread-safe'. [...]

I think it makes sense that only some functions need to be reentrant or thread-safe, and I have not enforced a requirement for thread-safety if a function is only used by a single thread. I also agree that it would be a good idea to document which functions are thread-safe or reentrant, so we don't forget and accidentally call unsafe functions from different threads concurrently, etc.

> The hope is that going through all of the functions and labeling them as one of these categories would make it clear whether we have race-conditions, potential deadlocks, etc. We would need to go through and ensure all of our driver and BSP functions are categorized appropriately.

If we are using this to look through the code for race-conditions and deadlocks, do you think it would be helpful if we modified the function names to make it obvious if they are reentrant/thread-safe? Maybe add an `_r` to the end for reentrant like in `strtok_r()`?

chaseblock, 2021-03-20:
Maybe. It certainly would make it easier to determine whether a function being called be an ISR is acceptable. On the other hand, we still may want to document thread-safety aside from reentrancy, which would either require another (non-standard) suffix, or a comment anyways.