lhr-solar/BPS #198: [RTOS] Design state machine for handling startup

SijWoo, 2020-09-12:
After the BPS faults due to low voltage, the next boot will automatically trip the system (opening the contactor). If we have a computer plugged in, during startup we should forcefully close the contactor so we can charge the battery to a point where it won't trip.

In the original loop during before anything is checked, there is a start up period that waits for an input from the keyboard to place the system in admin level to forcefully turn on the contactor. In this new rtos implementation, we have the ability to spawn and kill threads as we like. In the main loop instead of spawning all the necessary tasks, we should implement a state machine that spawns all threads for normal operation and spawn certain/different threads for admin operation.

ClarkPoon, 2020-09-19:
@SijWoo @alekxoo I think it would make sense to do this as part of the initialization task because that is where we do this in the non-RTOS version of the BPS. What do you think?

SijWoo, 2020-09-19:
That makes sense to me! Also I don't think I included this anywhere but we need to also check if the watchdog timer triggered previosuly

ClarkPoon, 2020-11-12:
After looking at this some more, I think it might make more sense to put this in the critical state thread than in the initialization thread because the critical state thread is what decides to turn on the contactor. I think it makes more sense for the initialization thread to only initialize the fault state sem4, critical state sem4, all of the other threads, and nothing else. What do you think @SijWoo

closed 2020-12-19 by ClarkPoon