lhr-solar/BPS #207: Look into recharging rates

chaseblock, 2020-09-26:
Potentially more related to MPPT, so tagging @dimembermatt on this. I think it is certainly reliant on BPS complicity though.

As a part of our recent discussions about proper management of the battery pack, it has come up that there exists proper charging procedure for these batteries, or at least there would exist a proper charging procedure if we had a dedicated, stable charger. Since we instead have an array, it may still be beneficial to look into how we should manage the charging of the batteries based on their existing voltage, charge, temperature, etc.

This is a part of the BMS functionality, and probably doesn't need to be addressed until after basic balancing (and related functionality) has been implemented and tested.

dimembermatt, 2020-09-26:
Updates on what the requirements should be?
I see potential to add functionality to the DC-DC converter algorithm to limit output based on battery voltage to prevent overcharging. Do I also get permission to read BPS CAN messages (I assume there are) for temperature probe data on the batteries? Or do I get a separate message to tell me to throttle charging output to X value and the BPS calculates how far I need to throttle to lower the temperature?

dimembermatt, 2020-09-27:
As a placeholder, I've implemented the following interface for the MPPT to regulate input power to the battery. Let me know if this changes. Closing [#19](https://github.com/lhr-solar/MPPT/issues/19).

[image]

Rationale for the implementation details is presented from the excerpt from the MPPT design document:
[image]

I'll check with Prof hallock to see if the physics for this checks out.

ClarkPoon, 2020-11-18:
I think we discussed this in a meeting over Zoom or Discord, but I am commenting here, so we have a record. 

> Potentially more related to MPPT, so tagging @dimembermatt on this. I think it is certainly reliant on BPS complicity though.
>
> As a part of our recent discussions about proper management of the [...]

My understanding is that the MPPTs know the total voltage of the battery pack, so the MPPTs should be able to handle the charge protocol without needing to receive any messages from the BPS, since the BPS will be balancing the module voltages. I think the purpose of the charging protocol is only to increase battery cycle life, so I am not sure if we would want to sacrifice power for this during the race. We might want to look into it for when we are charging before the race, but again, I don't think we care a lot about cycle life, since we are only planning on using this battery pack in one race.

> Updates on what the requirements should be?
> I see potential to add functionality to the DC-DC converter algorithm to limit output based on battery voltage to prevent overcharging. Do I also get [...]

I am guessing Array will need to know the BPS's shutoff voltage limits, current limits, and (optionally) the ideal charging protocol for the battery. I am not sure yet if you will need to know the BPS's temperature readings or temperature shutoff. I am hoping that the battery box will be designed in such a way that the BPS can keep the batteries cool enough, even when we are receiving maximum power from the array. 

Right now, the relevant BPS shutoff limits are:
Total pack Voltage (charging): 83.7 V to 124 V *
Total pack Voltage (discharging): 89.8 V to 124 V *
Max charging current: 20 Amps
Max discharging current: 100 Amps
Max temperature (charging): 40 C **
Max temperature (discharging): 55 C **
* Since we are monitoring Voltage on the module level, in practice, we may shutoff up to 1.5 V before we reach these limits.
** These temperature values will almost certainly be increased

@dimembermatt @chaseblock What are your thoughts on this?

dimembermatt, 2020-11-28:
I can probably work with just the voltage and current charge/discharge limits for now:

> Total pack Voltage (charging): 83.7 V to 124 V *
> Total pack Voltage (discharging): 89.8 V to 124 V *
> Max charging current: 20 Amps
> Max discharging current: 100 Amps

I'll have to read up on charging algorithms and/or protocols, otherwise it'll be a pretty primitive "I try to pump as much power given my constraints as possible all the time' for now in the DC-DC converter firmware.

closed 2022-04-26 by ClarkPoon