lhr-solar/BPS #268: BPS in unknown state after programming

chaseblock, 2020-12-14:
After programming the BPS leader board, the peripherals on the STM32F413 are not reset upon the software reset. This means, for instance, that if the BPS had the contactor enabled and is then flashed with a program that does not interface with the contactor (a test program perhaps), then the contactor will remain enabled. This also has potential effects for when the WDOG timer resets the chip (although this has not been tested yet).

The question is this: Is there a way to guarantee that after flashing or soft-resetting the board, all the peripherals are set to known and safe states?

chaseblock, 2020-12-14:
I should note that pressing the reset button on the board completely resets all peripherals.

ClarkPoon, 2020-12-16:
Can't we just solve this by:
1. On startup, (in fault state or critical state threads) call `BSP_Contactor_Off()`
2. In our standard operating procedures, require that the BPS is always reset after flashing new code

closed 2022-04-17 by manthanand