lhr-solar/BPS #378: Test CAN Task

ClarkPoon, 2021-06-26:
We need to run the test in Tests/Test_CAN_Task.c to verify that the CAN task works. Any issues that we run into when testing the CAN task should have an issue ticket opened or a response posted here

chaseblock, 2021-07-19:
When running this test, the board hardfaults at `os_core.c:422` after 6 calls to CANbus_BlockAndSend.

Stack trace from the hardfault:

```
>>> backtrace
#0  HardFault_Handler () at ../../Apps/Src/Task_FaultState.c:112
#1  <signal handler called>
#2  0x3f800000 in ?? ()
#3  0x0800493c in OSSched () at ../../RTOS/uCOS-III-STM32F4/uCOS-III/Source/os_core.c:422
#4  0x40400000 in ?? ()
```

Registers at the start of HardFault_Handler:

```
r0             0x0                 0
r1             0x10000000          268435456
r2             0x2000187c          536877180
r3             0x3                 3
r4             0x0                 0
r5             0x200065a6          536896934
r6             0x5                 5
r7             0x0                 0
r8             0x8080808           134744072
r9             0x9090909           151587081
r10            0x10101010          269488144
r11            0x11111111          286331153
r12            0x0                 0
sp             0x200011e8          0x200011e8 <OSCfg_ISRStk+396>
lr             0xfffffffd          -3
pc             0x8007430           0x8007430 <HardFault_Handler>
xPSR           0x1000003           16777219
fpscr          0x0                 0
msp            0x200011e8          0x200011e8 <OSCfg_ISRStk+396>
psp            0x20006508          0x20006508 <TaskSpam_Stk+840>
primask        0x0                 0
basepri        0x0                 0
faultmask      0x0                 0
control        0x0                 0
```

My guess is that we're corrupting a stack somewhere, or failing initialization at some step. Just a guess though.

chaseblock, 2021-07-20:
I'm going insane. Changing the line below in Test_CAN_Task.c seems to have solved the hardfault issue.

```c
-            CanData.f = (float)voltage/1000; //send data in volts
+            CanMsg.payload.data.f = voltage/1000.0; //send data in volts
```

In particular, it appears to be the change from `(float)voltage/1000` to `voltage/1000.0` that solved it (regardless of the assignment, the first faults).

I don't like unexplained faults, and the fact that this was coming from the OSSched function just makes me not like it more. The only (partial) explanation I can think of is that uC/OS is fucking with the floating point registers somehow, and that the compiler does slightly different things with this computation than the other one. Still, I don't know why this in particular was causing the issue, especially since there shouldn't be a context switch across it.

If anyone has any thoughts, I'd love to hear them.

@ClarkPoon @SijWoo @JimothyGreene

chaseblock, 2021-07-20:
This is certainly an issue with floating point and how it is interacting with uC/OS. I'll open a new issue to investigate.

closed 2021-08-29 by ClarkPoon