lhr-solar/BPS #430: Review MinionsASIC_Mutex usage

ClarkPoon, 2022-01-06:
Sometimes we are posting the mutex, then immediately pending it again, which might not be the most efficient way to handle this. We are also not using it sometimes when we communicate with the minion boards. Right now, I think the mutex is being used to protect access to the minion boards' SPI bus and to the minions asic array. We may want to use two mutexes for this

See:
https://github.com/lhr-solar/BPS/pull/271/files#r776791544
and
https://github.com/lhr-solar/BPS/pull/271/files#r776792239

manthanand, 2022-01-23:
I think the MinionsIO_Sem4 is supposed to be used for SPI bus protection.

ClarkPoon, 2022-01-23:
> I think the MinionsIO_Sem4 is supposed to be used for SPI bus protection.

The MinionsIO_Sem4 is for notifying us when we receive an SPI interrupt. It does not protect the SPI bus from multiple tasks accessing it at the same time.