lhr-solar/BPS #443: These issues are migrated from TODO list on drive

manthanand, 2022-04-18:
*set up system to make sure dashboard doesn't turn on motor during charging
*Verify charging state setup, what is required to charge

ClarkPoon, 2022-04-26:
A temporary solution that we are using is to just tell the driver to not turn on the motor if the BPS recently tripped due to an under-voltage, but it would be nice to have the system actually notify the driver about this (eg. via the BPS display board). 

We could also solve this by sending a message to controls that turning on the motor will fault the BPS, so they disable the motor, but this is not necessary (will just make it harder for the driver to accidentally/intentionally fault the BPS)

manthanand, 2023-02-13:
Closing this issue since we do not have a dashboard anymore.

closed 2023-02-13 by manthanand