lhr-solar/BPS #537: Remove Baremetal Functionality

manthanand, 2023-02-13:
At this point, the code base cannot run the bare metal code even if we compile for it. We should remove this functionality. This should be divided into multiple PRs with one PR corresponding to one library dependent on baremetal.

manthanand, 2023-02-19:
Examples of where Baremetal are used. 
[image]
[image]

MichaelMohn, 2023-04-01:
@manthanand Does this include removing references to it in comments

For example:
[image]
do I change this wording to not include baremetal?